iguanaworks/iguanair #10: check if windows is joining with threads correctly

IguanaBen, 2015-01-07:
**Reported by jdunn on 1 May 2012 01:20 UTC**
In the daemon.c file we have a call to joinThread, but there doesn't appear to be a similar call in the win32/service.c file.  My guess is that we're just not properly cleaning up the threads as they exit.  Not a big deal since this is one thread per device detected, but still.

IguanaBen, 2015-01-07:
**Commented by jdunn on 22 Jul 2013 21:31 UTC**
I ran taskmgr to check the thread count and plugged/unplugged the devices for quite a while.  Any increase in thread count dropped off once things had been idle for a bit.  Looks like we're cleaning up just fine.  I'll check again with a debugger but this is likely not a problem for some reason.

IguanaBen, 2015-01-07:
**Commented by jdunn on 29 Oct 2014 00:32 UTC**
If there is a leak it is slow, however, there does appear to be a more constant handle leak.  I appear to lose 4 handles per thread.  I'll need to look at the code fairly closely and ad some logging statements to track this down.  Not the end of the world by any means, but should be fixed.

jdunn14, 2017-02-25:
In windows we were not listening for clients threads to tell us when to join with them, leaking one thread handle per unplug.  We were also not closing three event handles per unplug event.  Finally, we were not releasing the client structure properly for connected clients during an unplug, leaking 1 file descriptor and one event handle.  All fixed on Windows in b562f45b776c74e64289804d608847996f98ed82 and 11bf93aa5153520e317d09cd6187e0114f960aa2.

closed 2017-02-25 by jdunn14