iguanaworks/iguanair #13: libusb device index is not the hardware device index

IguanaBen, 2015-01-07:
**Reported by jdunn on 18 Feb 2013 18:17 UTC**
Within the libusb based drivers we use a devIndex given by libusb and print it as if it were the hardware index.  It is not the hardware index and we need to find the hardware index instead/in addition to this index.

IguanaBen, 2015-01-07:
**Commented by bluey on 3 Nov 2014 01:28 UTC**
This may address #319 as well, so before closing this if you have a solution check and see if it fixes #319 as well.

jdunn14, 2017-03-10:
We have some patches from Yves Arrouye pertaining to Darwin that includes an OS X specific function to pull the device id and use it to provide an additional alias for each device.  We'll need to add code to do similar things on other hardware.  I also don't see an obvious reason to keep exposing the devIndex so we should just replace it w the HW index.

jdunn14, 2017-03-10:
The changes required for modern linux are quite simple, but things need to be checked on other platforms.

jdunn14, 2017-03-20:
Windows using libusbpre1 definitely does NOT work, but I'll have to try with a newer release.  Probably wouldn't be a bad idea to snap the version forward anyway.

jdunn14, 2017-03-28:
Seems to work fine with libusb-1.0 on OS X in my private repo, but Windows probably needs libusb 1.0 before I can say much there.

jdunn14, 2017-08-05:
Having tested on Windows I can confirm that the location is stable and dependent on the physical location under libusb-1.0-win32.  This will be fixed when we shift to that.  I'll wait to close this until the switch is complete.