iguanaworks/iguanair #14: Check for the kernel module

IguanaBen, 2015-01-07:
**Reported by jdunn on 18 Feb 2013 18:20 UTC**
Since there is now a kernel module available for our hardware and installed in Fedora18 we need to update the driver to identify when the module has claimed the USB device.  Once it is found we could provide a command line option to trigger a release of the hardware so that our daemon can communicate with it.

IguanaBen, 2015-01-07:
**Commented by stuart3 on 3 Mar 2013 14:59 UTC**
I just upgraded to kernel 3.8-trunk-amd64 (Debian experimental) and my 2 iguanaworks transceivers stopped working with lirc. I got these messages in /var/log/iguanaIR.log:

```
Mar 03 15:17:18 2013 ERROR: Is igdaemon already running?
Mar 03 15:17:18 2013 ERROR:   trying to claim usb:7:7
Mar 03 15:17:18 2013 ERROR:   updateDeviceList failed: Failed to set device configuration: Resource busy
Mar 03 15:17:18 2013 ERROR: scan failed.
Mar 03 15:24:49 2013 ERROR: Failed to read (interrupt end point): No such device
```

To get my transceivers working again I had to blacklist the iguanair kernel module

```
[pct] cat /etc/modprobe.d/iguanair-stuart.conf
blacklist iguanair
```

Is a command line option to ask the kernel module to release the transceiver (or blacklisting it) the only way to go? Why can't we use the iguanair module with licd? The module might get better performance.  /usr/bin/igdaemon wakes up quite often.

https://github.com/torvalds/linux/blob/master/drivers/media/rc/iguanair.c

IguanaBen, 2015-01-07:
**Commented by bluey on 3 Mar 2013 19:35 UTC**
I believe that the way to ask the kernel module to release the device is to unload the module with
```
sudo rmmod iguanair
```
but I think the general idea is to decide if you want to the user the kernel module, or our daemon. Then you disable our device or blacklist the kernel module respectively. The kernel module was written for use with the new "IR core" kernel functionality. "IR Core"is designed as a replacement for LIRC. To my knowledge no one is trying to get that it work with LIRC. Having said all that, we were not in any way involved with writing the kernel module and not yet decided if we will support it, or only support using our daemon (and blacklisting the kernel module).

IguanaBen, 2015-01-07:
**Commented by stuart3 on 11 Mar 2013 17:56 UTC**
Does "IR Core" really remplace lirc? Reading this web page it seems to be a different way to connect the transceiver to lircd.  http://wilsonet.com/?page_id=95

IguanaBen, 2015-01-07:
**Commented by bluey on 11 Mar 2013 18:17 UTC**
I'll have to look at this some more. My understanding is that part of the reason for ir-core is to be able to plug in a device and have it "just work" -- no config. I don't see how you can do this as a LIRC plugin as LIRC needs a config for the remote. I thought I remembered seeing some ir signal hex's be added to the kernel so it can decode some signals. But it does appear that LIRC can connect via ir-core. Now maybe there is a basic 'dumb lirc' in the kernel for simple things or you can use LIRC through ir-core for more complicated stuff. I'll keep this ticket open until we really look at ir-core and figure out what we want to do / not do with it. Also, there is some info in #285.

seanyoung, 2015-09-28:
ir-core configuration uses common linux infrastructure. You need the ir-keytable package installed (on Debian/Ubuntu, v4l-utils on Fedora). Keymaps are in /lib/udev/rc_keymaps/. You can select your keymap in /etc/rc_maps.cfg (maps rc driver to keymap, driver is called iguaunair). When the iguanair is plugged in, udev runs "ir-keytable -a rcX" and your remote will work as a normal input device.

Custom keymaps go in /etc/rc_keymaps/. You can test the device using "ir-keytable -t" There is no need for the lirc daemon and every application will take input from a remote as it is regular input device.

leamas, 2016-01-07:
> There is no need for the lirc daemon and every application will take input 
> from a remote as it is regular input device.

Wrong. The lircd daemon still has important functions even when it takes it's input from the kernel. It exposes a more advanced interface than the /dev/input/devices which is important in many usecases, notably some HTPC ones. 

That said, it is indeed possible to use remotes without LIRC.  But not without limitations.

leamas, 2016-01-07:
>. I'll keep this ticket open until we really look at ir-core and figure out what we want to 
> do / not do with it. Also, there is some info in #285.

From an LIRC perspective it would of course be better if you could get in touch with the kernel folks, and try to merge the added functionality in your user-space driver with the kernel driver. In the long run the kernel is probably the right place for this, but the problem is the sometimes complicated process of becoming part of the kernel team. A simple patch has taken me more than a year...

Bottom line: A unified kernel driver supported both by IguanaWorks and the current kernel maintainers would probably be the best long-term solution.  But, it's some work...

leamas, 2016-01-07:
> Does "IR Core" really replace lirc?
No

seanyoung, 2016-01-07:
> Wrong. The lircd daemon still has important functions even when it takes it's input from the kernel. It exposes a more advanced interface than the /dev/input/devices which is important in many [...]

What use cases? What limitations?

seanyoung, 2016-01-07:
> Does "IR Core" really replace lirc?
> 
> No
> 
> See: http://lirc.org/html/configuration-guide.html#overall-configuration-decisions

That doesn't explain anything. 

IR core (now called rc-core) was a way of upstreaming lirc, so yes it's intention was to replace lirc.

leamas, 2016-01-07:
Please read that link again. I don't want to quote here, unless necessary. It states several reasons when to use lirc, as well as the situation when there is no need for it.

seanyoung, 2016-01-07:
> That doesn't explain anything.
> 
> Please read that link again. I don't want to quote here, unless necessary. It states several reasons when to use lirc, as well as the situation when there is no need for it.

I can only find two reasons mentioned there. 
 1.  Your remote is not supported by rc-core
 2.  IR blasting 
However:
 1. You can add custom remotes yourself to rc-core and many remotes are already included
 2. IR blasting is supported by rc-core although user space tools could be improved

>     IR core (now called rc-core) was a way of upstreaming lirc,
> 
> No. It was a (very successful) way of upstreaming the LIRC kernel modules.

Yes, but those lirc modules were not accepted as-is, the kernel community wanted them to be tied to the input layer. So rc-core was created to bridge the gap and do in-kernel decoding. In the process the lirc daemon was bypassed and made obsolete. Some people keep on flogging a dead horse.

leamas, 2016-01-07:
> I can only find two reasons mentioned there. 
I find 7: 
  -  You might have a remote which is supported by LIRC but not the kernel.
  -  If you have a remote which isn't supported at all, LIRC is probably your best bet to get it running.
      > You can add custom remotes yourself to rc-core 
     Yep, But the workflow based on irrecord is still much more attractive.
  -  You might be on a non-Linux platform supporting lirc e. g., MacOS.
  -  You might have an application which is more or less designed to use LIRC.
  -  You might need LIRC's capabilities e. g., modes where a single remote button can be teached   to deliver different keys to the application. Handling input to multiple program is also easier with lirc
  -  You might want to send IR signals to other devices (IR blasting).
  > IR blasting is supported by rc-core although user space tools could be improved
  True. However, the design was made with the idea that lirc (i. e., irsend) is that tool.

  -   You might want to use lirc's applications e. g., irexec(1) which can run arbitrary commands in parallel with an application such as mythtv or xbmc.

>  So rc-core was created to bridge the gap and do in-kernel decoding. In 
> the process the lirc daemon was bypassed and made obsolete. 

No. In the process the /dev/lirc interface was kept intact with the very idea that lirc should have access to the raw, undecoded data. The lirc daemon is by no mean obsolete by this.  On the contrary, one of the design criterias  for ir-core was a close cooperation with the lircd daemon. 

On a sidenote, lirc of course uses the kernel decoding when suitable. From 0.9.4 this is the default setup.

jdunn14, 2017-03-29:
This really turned into a (now old) discussion that doesn't exactly address the original point of the ticket.  Going back to the original point: we do have a way to see what owns any device we have plugged in.  You can run igdaemon --devices and get a listing of what USB devices are recognized and what we can detect from the /sys filesystem about what if anything has claimed the device.

Closing this, but I'll be copying a couple of the responses to be merged into some HowTo documentation.

closed 2017-03-29 by jdunn14