iguanaworks/iguanair #15: WinLirc not reconnecting when igdaemon restarts

IguanaBen, 2015-01-07:
**Reported by jdunn on 16 Aug 2013 01:06 UTC**
WinLirc should reconnect if it's connection to the igdaemon is lost.  Unfortunately, this currently requires a change to the WinLirc source code due to the way libiguanair connects to client sockets directly.  See  #308 for some history.

IguanaBen, 2015-01-07:
**Commented by dukey on 15 Sep 2013 17:43 UTC**
When the system suspends and winlirc receives 
PBT_APMSUSPEND

It calls the deinit function on the plugin which shuts it down.

When the system resumes and PBT_APMRESUMEAUTOMATIC
is called, it waits 1second then calls the init function on the plugin. The iguana plugins fails with 

	connection	= iguanaConnect(deviceNumber);
	sendConnection	= iguanaConnect(deviceNumber);

	if (connection == INVALID_PIPE || sendConnection == INVALID_PIPE) {
		DPRINTF("connection == INVALID_PIPE || sendConnection == INVALID_PIPE\n");
		return false;
	}

I am not sure how to fix this.

IguanaBen, 2015-01-07:
**Commented by jdunn on 16 Sep 2013 18:38 UTC**
I have some ideas of things to try and will post modified binaries if I managed to get them working.  Just to log this somewhere other than my email: we should try registering our service for windows power management events to possibly rescan the usb bus earlier.

IguanaBen, 2015-01-07:
**Commented by dukey on 16 Sep 2013 18:41 UTC**
Can't you just rescan if a users tries to connect ?

jdunn14, 2017-07-26:
We technically could not rescan when the user tried to connect, but should look into this again with our newly working ctl interface.  We might be able to replace a DLL in WinLIRC and get reconnects.

jdunn14, 2017-08-01:
I've written code to make this work.  It requires building the iguanaIR.dll library with VS2008, but the Express version works fine.  I'll include a copy of this DLL in our windows packaging in this coming release.

closed 2017-08-01 by jdunn14