iguanaworks/iguanair #16: New Resource bush error message

IguanaBen, 2015-01-07:
**Reported by bluey on 1 Oct 2013 19:57 UTC**
If you start igdaemon and it cannot connect to the device, it reports
```
Oct 01 18:50:49 2013 ERROR:   updateDeviceList failed: Failed to set device configuration: Resource busy
```

Can we make this ask if igdaemon is already running or if the kernel driver iguanair is not blacklisted? It would be nice to point users in the right direction to fix / and we can have a FAQ entry on this error.

IguanaBen, 2015-01-07:
**Commented by jdunn on 2 Oct 2013 00:15 UTC**
I tried to replicate the problem and found that we actually already have an appropriate error message.  To replicate the error message turn on the iguanair kernel driver and then fire up igdaemon.  This is what happens:
```
jdunn@margarita usb_ir$ sudo igdaemon -nvvv
Oct 01 20:03:23 2013 DEBUG: Parameters:
Oct 01 20:03:23 2013 DEBUG:   recvTimeout: 1000
Oct 01 20:03:23 2013 DEBUG:   sendTimeout: 1000
Oct 01 20:03:23 2013 DEBUG:   driverDir: /usr/lib64/iguanaIR
Oct 01 20:03:23 2013 INFO: Loaded driver: /usr/lib64/iguanaIR/libusb.so
Oct 01 20:03:23 2013 ERROR:   updateDeviceList failed: Failed to set device configuration: Resource busy
Oct 01 20:03:23 2013 ERROR: Check device status with igdaemon --devices
```

Following the directions from that last line:
```
jdunn@margarita usb_ir$ sudo igdaemon --devices
Detected Iguanaworks devices:
  USB IR device number 4 on bus 4:
    claimed by kernel driver 'iguanair'
```

Seems pretty good to me.  This will also print out a message if the device is claimed by another instance of the igdaemon.  Closing this one.

IguanaBen, 2015-01-07:
**Commented by bluey on 11 Oct 2013 15:26 UTC**
I'm reopening it the ticket appears to be valid, but only on Raspberry Pi machines. See #320 and top of #314.

IguanaBen, 2015-01-07:
**Commented by bluey on 3 Nov 2014 00:45 UTC**
Confirming issue on Raspberry Pi. When device plugging in and used by kernel, I get:

```
igdaemon -nvvv --devices
Nov 02 17:42:22 2014 DEBUG: Parameters:
Nov 02 17:42:22 2014 DEBUG:   recvTimeout: 1000
Nov 02 17:42:22 2014 DEBUG:   sendTimeout: 1000
Nov 02 17:42:22 2014 DEBUG:   driverDir: /usr/lib/iguanaIR
Nov 02 17:42:22 2014 INFO: Loaded driver: /usr/lib/iguanaIR/libusb.so
Detected Iguanaworks devices:
  USB IR device number 4 on bus 1:
```

but despite seeing a device, it cannot connect as the resource is busy:

```
sudo igdaemon -nvvv
Nov 02 17:43:00 2014 DEBUG: Parameters:
Nov 02 17:43:00 2014 DEBUG:   recvTimeout: 1000
Nov 02 17:43:00 2014 DEBUG:   sendTimeout: 1000
Nov 02 17:43:00 2014 DEBUG:   driverDir: /usr/lib/iguanaIR
Nov 02 17:43:00 2014 INFO: Loaded driver: /usr/lib/iguanaIR/libusb.so
Nov 02 17:43:00 2014 ERROR:   updateDeviceList failed: Failed to set device configuration: Resource busy
Nov 02 17:43:00 2014 ERROR: Check device status with igdaemon --devices

```

IguanaBen, 2015-01-07:
**Commented by bluey on 3 Nov 2014 00:47 UTC**
Also:

```
lsmod|grep ig
iguanair                5914  0 
rc_core                17576  12 lirc_dev,ir_lirc_codec,ir_rc5_decoder,ir_nec_decoder,ir_sony_decoder,ir_mce_kbd_decoder,ir_jvc_decoder,ir_rc6_decoder,ir_sanyo_decoder,rc_rc6_mce,iguanair
```

IguanaBen, 2015-01-07:
**Commented by bluey on 3 Nov 2014 01:02 UTC**
Probably this is really #296.

jdunn14, 2017-03-21:
Tested on RP and I agree we're missing a line of text.  There should be a "claimed by kernel driver 'iguanair'" line that isn't getting printed at least on RP.

jdunn14, 2017-03-22:
Fixed in 275c23bde1ac8b7cdb52a1656e60e0fe5e87326d.  Closing this ticket out and the fix will be in the public repo in the next couple weeks.

closed 2017-03-22 by jdunn14