iguanaworks/iguanair #7: IG_DEV_SETCHANNELS with firmware revision 0x0003

IguanaBen, 2015-01-07:
**Reported by jdunn on 29 Jul 2010 22:46 UTC**
So, looking at the igdaemon log I think it may be a bug in the daemon.  The telling lines are:

```
Jul 29 01:17:10 2010 ERROR: Unknown packet type in request: 0x17
Jul 29 01:17:10 2010 ERROR: handleClientRequest(0x00) failed with: 22 (Invalid argument)  
```

Basically it looks like the daemon didn't translate the client's
IG_DEV_SETCHANNELS call as it was supposed to according to
protocol-versions.c.  Upgrading the firmware is a fine work-around for the problem.  I would guess that a breakpoint in translate device's if statement would find the issue fairly fast.

```
bool translateDevice(uint8_t *code, uint16_t version, bool fromDevice)
{
    if (version <= 4)
        version = 0;
    else
        version = 1;

    return translateClient(code, version, fromDevice);
}
```

IguanaBen, 2015-01-07:
**Commented by jdunn on 22 Jul 2013 21:10 UTC**
This is still a bug, and here's some igdaemon output to go with it based on running:
```
igclient --set-channels 0x3
```

Connected to version 0x306
```
Jul 22 17:08:02 2013 INFO: Found client using protocol version 1
Jul 22 17:08:02 2013 INFO: Request handled within daemon: 0xfe
Jul 22 17:08:02 2013 DEBUG3: Successfully wrote packet: 0xfe00000002000000
0x0100
Jul 22 17:08:02 2013 INFO: Request handled within daemon: 0x10
Jul 22 17:08:02 2013 DEBUG3: Successfully wrote packet: 0x1000000001000000
0x02
Jul 22 17:08:02 2013 INFO: Request handled within daemon: 0x17
Jul 22 17:08:02 2013 DEBUG3: Successfully wrote packet: 0x1700000001000000
0x03
```

Using version 0x0003:
```
Jul 22 17:08:48 2013 INFO: Found client using protocol version 1
Jul 22 17:08:48 2013 INFO: Request handled within daemon: 0xfe
Jul 22 17:08:48 2013 DEBUG3: Successfully wrote packet: 0xfe00000002000000
0x0100
Jul 22 17:08:48 2013 ERROR: Unknown packet type in request: 0x10
Jul 22 17:08:48 2013 ERROR: handleClientRequest(0x00) failed with: 22 (Invalid argument)
Jul 22 17:08:48 2013 DEBUG3: Successfully wrote packet: 0x00000000eaffffff
Jul 22 17:08:48 2013 ERROR: Unknown packet type in request: 0x17
Jul 22 17:08:48 2013 ERROR: handleClientRequest(0x00) failed with: 22 (Invalid argument)
Jul 22 17:08:48 2013 DEBUG3: Successfully wrote packet: 0x00000000eaffffff
```

The protocol may supposed to be different, but there is little chance that eaffffff is the correct value to send.  It's an old firmware revision that probably isn't in the wild anymore so this is low priority, but it is a daemon bug.

IguanaBen, 2015-01-07:
**Commented by jdunn on 26 Oct 2014 02:24 UTC**
Well, the bug is larger than expected.  Basically the translateDevice call needs to be made before checkIncomingProtocol.  The protocol being passed from the client.c code will always be the newest, but when checking whether it's valid we need to check using the translated code, not the incoming, since it will be translated before it goes out.  This whole section of the code should be looked at more closely, but not tonight for this ticket.  Further, the protocols have not changed meaningfully since the body/loader break years ago so this is very low priority, but should be fixed eventually.  Also, the 0x10 is a feature check being triggered by client.c code but SHOULD be handled in the daemon instead, and not triggered for this device version.

jdunn14, 2017-08-04:
I've reworked this code and think I have a handle on it.  The codes are properly translated before sending them out to the device.  This is actually done after checkIncomingProtocol because it turns out we were missing a protocol definition so it can work in this order.  I believe we can close this one based on what I'm about to commit.

commit 446a0807 (2017-08-05): In preparation for cleaning up the protocol I've revisited the protocol translation code and fixed #7 along the way.
closed 2017-08-04 by jdunn14