iguanaworks/iguanair #8: take a careful look at the lirc driver

IguanaBen, 2015-01-07:
**Reported by jdunn on 23 Nov 2010 19:22 UTC**
I believe the lirc driver could probably use a careful check-through to look for a few specific thigns:
 - check if we need to do anything to support SEND_START and SEND_STOP
 - see if we can get the timings any more precise for LIRC
  - possibly look into using the compression schemes we've come up with
 - overall just make sure it's not unnecessarily complicated or delaying.

jdunn14, 2017-03-08:
Don't forget to check for re-connect on igdaemon restart.

jdunn14, 2017-07-24:
Look taken.  Mostly fixed up although I'm saving a few questions to ask leamas.  Send start/stop seem to be working fine.  Overall code seems to be working.  I'm not sure what we can improve on the timings, but I'll check on the gap timing.

closed 2017-07-24 by jdunn14