iguanaworks/iguanair #9: igdaemon.exe stops when data is not consumed by client

IguanaBen, 2015-01-07:
**Reported by mortenab on 13 Dec 2010 10:36 UTC**
If data is not consumed by client, the igdaemon.exe crashes. 

Reproduce by:
 * Make a program that sends 0x12 command to start receiving IR data
 * Do not consume data from igdaemon (iguanaIR-0 named pipe)

IguanaBen, 2015-01-07:
**Commented by jdunn on 16 Dec 2010 12:15 UTC**
Thank you for the trio of bug reports.  I just wanted to get a couple more details about them.  First of all, I assume these were all triggered on the same system, so which version of Windows are you running and is it 32 or 64 bit.  For this specific ticket, when does it crash?  Right away or does it grow in memory use then crash?

IguanaBen, 2015-01-07:
**Commented by mortenab on 16 Dec 2010 12:25 UTC**
Windows 7 64-bit. 

The program crashes after a couple of seconds after I stop consuming data (when I stop in a breakpoint in VS). I haven't watched memory usage. But it is persistant.

IguanaBen, 2015-01-07:
**Commented by mortenab on 17 Dec 2010 08:03 UTC**
It does not crash..  It just stops sending (ir-)data to the client.  A restart of igdaemon corrects the problem.

IguanaBen, 2015-01-07:
**Commented by jdunn on 29 Oct 2014 00:48 UTC**
Looks like this is still a problem.  I suspect that the main thread is stuck when it tries to write data back to the hung client and then never properly services other clients.  This would not be triggerable with our igclient, but custom code could cause it so it needs to be fixed.  I bet we can cause the same thing on Linux so I'll have to check it there.

jdunn14, 2017-03-19:
Looks like it works fine in Linux.  The non-blocking code that makes it work was (re-)enabled way back in Dec 2008.  I'll need to check this on Windows, but it seems like it should not be a problem.

jdunn14, 2017-03-20:
Well, it is definitely still a problem on Windows, but likely only on Windows.

jdunn14, 2017-08-05:
Checking in windows if there is a memory leak it is so slow as to be meaningless.  The hardware does eventually crap out as well, possibly because the main thread is blocked.

jdunn14, 2017-08-05:
Fixed in a branch of my local repo.  Closing this out.

closed 2017-08-05 by jdunn14