iguanaworks/iguanair #20: C# wrapper

Eonasdan, 2015-07-24:
I have a project in C# using the IR blaster. Currently this relies on LIRC and Mono, which I would very much like to remove.

Unfortunately, I don't know much about C++ or building drivers. I know you built a plugin for WinLIRC. Do you have, can you make, can you help me make a C# wrapper for your driver?

Ultimately, I'd love to see a driver (and wrapper) for Windows 10 IoT.

Thanks

jdunn14, 2015-07-28:
At the moment we do not provide anything specific to C#.  There has been talk of adding a COM layer for portability across languages on Windows, but we haven't made any real progress there at this time.  It may be on the radar for the beginning of 2016.  In the mean time, I've never done it, but I believe there's a way to call into native DLLs from C#.  A quick google search shows that C# has a PInvoke mechanism for such things.  Please take a look at that process and if you have specific questions I'll be happy to answer them as they come up.

jdunn14, 2015-07-28:
Almost forgot, I will be testing our current install package under Windows 10 in the couple month or two and don't foresee any major hurdles.  As for Win10 IoT support, I've never looked at it before, but I expect it would be a similar time frame to the COM or similar support.

Eonasdan, 2015-08-02:
I'd love to help write a C# wrapper, but I can't even get the project to build :(

VS 2015 gives `Macro definition of snprintf conflicts with Standard Library function declaration` when trying to build `ALL_BUILD` after what I think I've done correctly with the CMake program.

Looks like 2015 includes `snprintf` in `stuido.h`.

Eonasdan, 2015-08-03:
I managed to fix the `snprintf` error by commenting out these two lines.

[image]

I tried to get Swig to generate a wrapper based on the interface file you have but must not have done it correctly.

I'm not getting this error:

` This is likely because the managed PInvoke signature does not match the unmanaged target signature.`

Swig generated this code that I've tried to get to work, but again my understanding of C++ is limited.

```
public static int connect(string name) {
    int ret = IguanaIrpinvoke.connect(name);
    return ret;
  }

[DllImport(DllPath, EntryPoint = "iguanaConnect_real")] //got this from DLL dump
    public static extern int connect(string name);
```

This project is to update my prototype that I currently have working with Linx, LIRC and your blaster. I'm trying to get this to work without LIRC on Win 10 IoT. If I can make it work, it could result in my company ordering a whole bunch of your blasters.

jdunn14, 2015-08-03:
Well, we've never built the project with VS2015 so I'm not terribly surprised that it doesn't build out of the box.  I'll have to look at why, but will probably not be able to do some directly until next week.  I'll make the #defines that you commented out dependent on the existence of the symbol or the version of the MSVC compiler.

All that said, I'm not quite sure why you need to build the project in order to write the wrapper.  Most higher level languages, of which I'm most comfortable with Python, provide a mechanism to take an already built DLL and execute functions within that DLL.  The swig approach we used to provide Python bindings should not be necessary, and building them seems unnecessary for your task.  If you disable the Python / SWIG parts of the CMakeLists, most likely by just making sure that cmake cannot find either one, then you should be able to build the DLL.

Even without rebuilding the DLL you should be able to load the DLL we distribute using PInvoke and make calls into it.   There's a tutorial of the basics of the method here:
https://msdn.microsoft.com/en-us/library/aa288468(v=vs.71).aspx

See if you can install our package, load the DLL using DllImport and then call the iguanaConnect function.  It will return

Eonasdan, 2015-08-03:
According to the `dumpbin` utility, these are the exported functions from `iguanaIR.dll`. Using `iguanaConnect` gives me: `Unable to find an entry point named 'iguanaConnect'`

```
1    0 0000109B appendHex = @ILT+150(_appendHex)
          2    1 0000104B iguanaClose = @ILT+70(_iguanaClose)
          3    2 00001028 iguanaCode = @ILT+35(_iguanaCode)
          4    3 000012A8 iguanaConnect_internal = @ILT+675(_iguanaConnect_internal)
          5    4 000012B7 iguanaConnect_real = @ILT+690(_iguanaConnect_real)
          6    5 000010AF iguanaCreateRequest = @ILT+170(_iguanaCreateRequest)
          7    6 0000103C iguanaDataToPinSpec = @ILT+55(_iguanaDataToPinSpec)
          8    7 00001172 iguanaFreePacket = @ILT+365(_iguanaFreePacket)
          9    8 00001271 iguanaPinSpecToData = @ILT+620(_iguanaPinSpecToData)
         10    9 00001032 iguanaReadBlockFile = @ILT+45(_iguanaReadBlockFile)
         11    A 00001037 iguanaReadPulseFile = @ILT+50(_iguanaReadPulseFile)
         12    B 0000100A iguanaReadResponse = @ILT+5(_iguanaReadResponse)
         13    C 000010B9 iguanaRemoveData = @ILT+180(_iguanaRemoveData)
         14    D 00001078 iguanaResponseIsError = @ILT+115(_iguanaResponseIsError)
         15    E 000012BC iguanaWriteRequest = @ILT+695(_iguanaWriteRequest)
         16    F 000011E5 message = @ILT+480(_message)
         17   10 00001285 wouldOutput = @ILT+640(_wouldOutput)
```

using

```
[DllImport(DllPath, EntryPoint = "iguanaConnect_real")]
public static extern IntPtr connect(string name);
```

Still gives me a mismatch signature error

jdunn14, 2015-08-03:
In the iguanaIR.h header the definition of iguanaConnect_real is:
```
PIPE_PTR iguanaConnect_real(const char *name, unsigned int protocol);
```

I think the code should be something like:
```
[DllImport(DllPath, EntryPoint = "iguanaConnect_real")]
public static extern IntPtr connect(string name, uint protocol);
```

You should be able to get the proper value for protocol when calling connect by looking in the same header for IG_PROTOCOL_VERSION, but at this point it's just 1.  If the code above doesn't complain about entry points try calling:
```
connect("0", 1);
```

By the way, this connect call is the only one that we had to do this odd _real vs the commonly used name.  The other functions should be more straight forward.  Let me know how this goes and we'll fight through this or the next problem to get you a working C# wrapper.

jdunn14, 2017-07-30:
Did anything ever come of this?  I'm interested in picking up from wherever it was left.

jdunn14, 2017-08-25:
Closing this ticket, but I started making a C# wrapper myself using Mono.  So far I can connect to the device and trigger a GETVERSION, but I have not tried to unpack the returned data yet.

closed 2017-08-25 by jdunn14