iguanaworks/iguanair #21: Can't compile for OSX

jgimenez, 2015-11-14:
When doing a `cmake -G Xcode` I get an error:

```
-- The C compiler identification is Clang 7.0.0
-- Check for working C compiler using: Xcode
-- Check for working C compiler using: Xcode -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
CMake Error at CMakeLists.txt:92 (message):
  Unrecognized CMAKE_SYSTEM_NAME

-- Configuring incomplete, errors occurred!
```

Looking at the code history, I see there used to be a Makefile but now you're using CMake. Is it possible that OSX was forgotten in this transition? How can I proceed? Thanks

jdunn14, 2017-03-03:
I don't know if this is useful to you this late, but the most recent commits allow for building under macports on OS X.

closed 2017-03-03 by jdunn14

jgimenez, 2017-03-14:
Better late than never, they say!
I've been able to make it compile, thanks. Will test later with the device.

jdunn14, 2017-08-02:
By the way, there's a rather annoying firmware bug that is only evident on OS X, but there's a workaround in the github repo.  We're trying to find a better fix, but I can send and receive signals on OS X.