iguanaworks/iguanair #24: any alternative to WinLIRC's transmit.exe?

wmclaxton, 2016-08-16:
I am using IguanaWorks USB device and WinLIRC 0.9.0i to communicate with two set-top boxes for a large broadcast recording project.  The recording application is NextPVR, which supports a command line execution to implement channel changes via an IR emitter, whenever the user switches channels during viewing or recording.  We are testing this using transmit.exe and setchannel.bat, a utility I wrote to send the full channel number via transmit.exe.

Both setchannel.bat and transmit.exe work at the command line, but both fail when invoked from within NextPVR.

Apparently, the problem is, the transmit.exe source code (https://sourceforge.net/p/winlirc/code/HEAD/tree/trunk/winlirc/transmit.cpp) is using WP_COPYDATA.  This assumes a client/service architecture, and uses the clipboard (WM_COPYDATA) to communicate between client and server.  Since client and server run under different desktop sessions, they don't share a clipboard so the messaging fails.

Do you have any suggestion for how this simple code could be rewritten to avoid this problem?  Can the IGdaemon be used to replace transmit.exe?  Do you know of developers who may have created alternatives to transmit.exe which do not use clipboard communications?

jdunn14, 2016-08-16:
I can give you a couple of options for bypassing WinLIRC for transmission.  First, we distribute an executable as part of our driver package called igclient.  You can generate the space/pulse lengths for your signal either by recording or generating the listing based on decoding the lircd.conf.  The required format for igclient looks like:

pulse: 8000
space: 4000
pulse: 500
space: 500
pulse: 500
space: 1000
pulse: 500
etc.

The file should start and end in pulses and just encodes a single button.  Once you have that file try:
igclient --send filename.txt

That application will read the contents of the text file, connect to the igdaemon over a windows named pipe, and send the data that way.  Alternatively, if you want to work with your own code we provide a .h file and a .lib file so that you can write code that uses the same API as the igclient.  If you would prefer to got this route let me know and I can provide some example C code, but basically you call into the DLL to connect to the igdaemon, create a request, append the pulse/space lengths, and send the request to the igdaemon.

If you have a problem generating the send files you need send me the WinLIRC configuration file and I'll see what i can do.

-Joseph

jdunn14, 2017-02-21:
Since there was never a response I'm going to close this ticket.  Hope it's all working for you now!

closed 2017-02-21 by jdunn14

wmclaxton, 2017-02-21:
In case someone comes across this and is looking for a solution, here you go.

1.  We created a modified version of WinLIRC transmit.exe which implements socket communications (so it's called 'stransmit.exe').  This enables channel selection requests to be sent from any 3rd party application (eg- NextPVR).  The original transmit.exe relied on Windows clipboard to pass the data, which wouldn't work from within a 3rd party application (since they run under different Windows accounts).

2.  We also wrote a batch script that takes care of waking WinLIRC, setting the IR transmission port (via the IguanaWorks IR transciever) and sending a complete channel change request as a series of individual key presses.  Setting the transmission port is required if you are communicating with multiple devices.

Write to me wmclaxton [at] gmail.com if you would like the source files at no cost.  Happy to share them.

lamb0985, 2023-02-18:
i know this is so old, but could you send me the stransmit.exe and/or batch script? I am also struggling to get my iguana blaster to work with winlirc so that i can use it with nextpvr to change channels on my set top box. my email is lamb0985 [at] gmail.com