iguanaworks/iguanair #25: weird SONAME and  .so links.

leamas, 2017-01-25:
The generated SONAME looks unversioned: 

     $ objdump -p foo/usr/lib/libiguanaIR.so | grep SONAME
     SONAME               libiguanaIR.so

A SONAME is supposed to be versioned. Also. the standard setup on POSIX systems is symlinks corresponding to the SONAME.  Compare e. g., llibgmp:

      $ objdump -p /usr/lib64/libgmp.so.10.3.1 | grep SONAME
      SONAME               libgmp.so.10
     [mk@snorken iguanaIR]$ ls -l /usr/lib64/libgmp*.so*
     lrwxrwxrwx 1 root root     16 Jun 22  2016 /usr/lib64/libgmp.so.10 -> libgmp.so.10.3.1
     -rwxr-xr-x 1 root root 604888 Jun 22  2016 /usr/lib64/libgmp.so.10.3.1

This is how programs and in particular distributions handles the binary compatibility. Please make update the iguanaIR soname to a versioned one and update the installed files. BTW, this is a breakage introduced with cmake, the autotools libraries were fine.

leamas, 2017-01-25:
A simple-minded patch below. This is unconditional and thus probably not acceptable as-is:
    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index f8c37d1..59d92e6 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -99,6 +99,9 @@ add_library(iguanaIR SHARED
                          ${BASESRC})
     set_property(TARGET iguanaIR
                  APPEND PROPERTY COMPILE_DEFINITIONS IGUANAIR_EXPORTS)
    +set_property(TARGET iguanaIR PROPERTY VERSION "0.0.0")
    +set_property(TARGET iguanaIR PROPERTY SOVERSION 0)
    +
     install(TARGETS iguanaIR
             DESTINATION ${LIBDIR})
 
    --
    2.9.3

which sets up the desired result:
    $ ls -l foo/usr/lib/libiguanaIR.so*
    lrwxrwxrwx 1 mk mk    16 Jan 25 22:08 foo/usr/lib/libiguanaIR.so -> libiguanaIR.so.0
    lrwxrwxrwx 1 mk mk    20 Jan 25 22:08 foo/usr/lib/libiguanaIR.so.0 -> libiguanaIR.so.0.0.0
    -rwxr-xr-x 1 mk mk 58512 Jan 25 22:07 foo/usr/lib/libiguanaIR.so.0.0.0
    [mk@snorken build]$ objdump -p foo/usr/lib/libiguanaIR.so.0.0.0  | grep SONAME
      SONAME               libiguanaIR.so.0

jdunn14, 2017-02-18:
Believe this is fixed by the merge of #28

leamas, 2017-02-20:
Yes. This sets the so-version to 0.0.3, which was the last version generated by autoconf.  This shouls keep any distros packaging this happy, avoiding a son-name bump.

On fedora, the combination of the old circular lirc <-> itguanair dependency  combined with the bad so-name bump created an insane mess :(. Perhaps, others could avoid this.

Closing

closed 2017-02-20 by leamas