iguanaworks/iguanair #26: Compilation errors in swig-generated code.

leamas, 2017-01-25:
Compiling gives errors as attachhed. They seem bad enough to drop the concerned python code from the bulid.

[warnings.txt](https://github.com/iguanaworks/iguanair/files/730775/warnings.txt)

leamas, 2017-01-27:
BTW, this is on the 1.1.0 release. Building the github HEAD does *not* have this issue

jdunn14, 2017-02-18:
I just checked the current HEAD and agree that it appears to be at least compiling there although I haven't done any testing beyond that yet.  I would say we're past due for a new release, so we'll confirm HEAD is doing what it should and put out an official 1.1.2 or possibly 1.2.0 once the current packaging changes are ready and I've had a chance to look through some internal TODOs.

leamas, 2017-02-20:
OK, makes sense. Closing

closed 2017-02-20 by leamas