iguanaworks/iguanair #29: No systemd support.

leamas, 2017-01-27:
Effectively, all linux:es have moved to systemd. Adding systemd support is trivial, and doesn't hurt if system uses traditional sysV.

Attaching  /lib/systemd/system/iguanair,service and  usr/lib/tmpfiles.d/iguanair.conf form the fedora package. They should be fine upstream, one of the ideas with systemd is that all distributions shares these files. txt-suffuixes as required by github...

Remaining  issue is to configure  80-iguanaIR.rules so that it doesn't duplicate creation of the temprary directory on a systemd installation.

[iguanaIR.service.txt](https://github.com/iguanaworks/iguanair/files/736307/iguanaIR.service.txt)
[iguanair.conf.txt](https://github.com/iguanaworks/iguanair/files/736309/iguanair.conf.txt)

jdunn14, 2017-02-18:
Should be fixed by the merge of #28

leamas, 2017-02-20:
Indeed, closing

closed 2017-02-20 by leamas