iguanaworks/iguanair #35: "fatal error: config.h: No such file or directory" on attempt to make

eugenet8k, 2017-02-26:
Hi, I am new to linux specifics, I just bough your IR USB transiver device and trying to make it work with Raspberry Pi 3.  It didn't work with default apt-get package `lircmd 0.9.0-pre1`, because there is no iguana driver, so I built the last version of Lirc from git `lircmd 0.9.4d`, which doesn't have driver as well, so I am trying to build this project now. Although I am getting error on attempt to `make`:

```
pi@raspberrypi:~/git/iguanair/software/lirc-drv-iguanair $ make
cc -I../usb_ir -fpic -DPLUGINDOCS=\"/usr/local/share/doc/lirc/plugindocs\" -I/usr/local/include    -c -o iguanair.o iguanair.c
In file included from /usr/local/include/lirc/ir_remote_types.h:52:0,
                 from /usr/local/include/lirc_driver.h:21,
                 from iguanair.c:31:
/usr/local/include/lirc/include/media/lirc.h:9:20: fatal error: config.h: No such file or directory
 #include <config.h>
                    ^
compilation terminated.
<builtin>: recipe for target 'iguanair.o' failed
make: *** [iguanair.o] Error 1
```

Any help would be greatly appreciated!

jdunn14, 2017-03-05:
When I ran into that problem earlier I was able to comment out the offending line in the lirc.h header.  See if that lets you get past this point in the build.

leamas, 2017-03-19:
This is an upstream error,  fixed in master but not released.See https://sourceforge.net/p/lirc/tickets/250

jdunn14, 2017-03-19:
Good to hear.  I'll leave this here in case any of our customers runs into it, but let us know when the fixed version is released and we'll close this ticket.

leamas, 2017-03-19:
My bad. It's actually already  fixed in 0.9.4d (end of January).  Bur you might as well leave thsi open for some time, not all updates lirc to latest version...

jdunn14, 2017-07-28:
I also have a work around in the lirc-drv-iguanair Makefile that should work on at least some platforms.  Overall I'm going to call this fixed for now.

closed 2017-07-28 by jdunn14

adamcstephens, 2017-08-27:
As of 983b0559c09db3228e3ed05c58cde1e15f3273d7 I still had to comment out the config.h line.

```
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

leamas, 2017-08-28:
This might be https://sourceforge.net/p/lirc/tickets/300/, fixed in the last debian and fedora packages, but not in the latest upstream release (it's on the release-0.10 branch, though).