iguanaworks/iguanair #38: lirc driver does not support device enumeration

leamas, 2017-03-29:
As heading says, the lirc driver lacks support for device enumeration. LIRC implements enumeration using the DRVCTL_GET_DEVICES drvctl.

The easy way to check is to use `mode2 --list-devices`

Enumerating the iguana  devices should give something like
  
    /run/iguanaIR/0   [1234:5678]   /sys/class.......
    /run/iguanaIR/1  [1234:6789]   /sys/class....

The only mandatory part is the first word i. e., the actual device to use in LIRC, The rest is optional,  free-format info usable in a UI.

I'm attachi
[0001-Add-device-enumeration.patch.zip](https://github.com/iguanaworks/iguanair/files/878913/0001-Add-device-enumeration.patch.zip)

ng a patch which fixes the generic parts, without actually doing the enumeration

jdunn14, 2017-03-29:
Great timing on this ticket.  I've been working in a private repo until I get a everything pretty much together for the new release.  Yesterday I added a function to the iguanaIR API to enumerate devices.  I'll apply this patch in there and hook it up to the new enumeration code.  The enumeration mechanism is just a directory listing on Linux, but that will be replaced w talking to igdaemon on all platforms in the following release.

By the way, if you have other features/patches against the current repo now would be a good time to pass them along so I have to deal w the merge instead of you.

leamas, 2017-03-29:
> By the way, if you have other features/patches against the current repo now would be a good time to pass them along so I have to deal w the merge instead of you.

No, this should be the last one ;)

leamas, 2017-03-29:
> The enumeration mechanism is just a directory listing on Linux, but that will be replaced w talking to igdaemon on all platforms in the following release.

In this case you could just keep the current code using ` drv_enum_glob()`, possibly adjusting the glob pattern.

jdunn14, 2017-03-29:
By the way, up to three names might actually refer to the same device:
 - the device number (0, 1, 2....)
 - a user-defined alias (livingroom, tv, etc.)
 - another alias based on the USB location (1-3, 2-4, etc.) *this one is new to the next release

Is there was proper way to indicate this through the function call?  Even if we could just flag the 2 aliases as such that might be good.  I doubt people will be confused when 2-3 devices show up per physical device and all mean the same thing, but you never know.

jdunn14, 2017-03-29:
Perfect.  I'll add some symlink reading and return lines a lot like what you have there.

jdunn14, 2017-07-24:
Proper device enumeration is added to my local repository.  Will discuss a couple questions w leamas shortly, but closing this as done for now.

closed 2017-07-24 by jdunn14