iguanaworks/iguanair #39: irsend: hardware does not support sending

brianjmurrell, 2017-04-27:
Hi.  I am trying to debug my installation of one of [these](http://www.iguanaworks.net/wiki/lib/exe/fetch.php?tok=32802f&media=https%3A%2F%2Fwww.iguanaworks.net%2Fwp-content%2Fuploads%2Fproducts%2Fusb-ir-transceiver%2FUSB-LED-300x113.jpg).

I've tried to follow along these [instructions](http://www.iguanaworks.net/wiki/doku.php?id=usbir:gettingstarted) but am getting nowhere.  I am using the device as a transmitter not a receiver.  This was all working in the past but just stopped for whatever reason.

I have your repo installed.  I have rebuilt lirc on Ubuntu and installed the .deb.  Here is the current state:

```
# ps -ef |grep igdae 
iguanair  6001     1  0 08:46 ?        00:00:00 /usr/bin/igdaemon --log-level=0 --send-timeout=1000 --receive-timeout=1000 --driver=libusb -l /var/log/iguanaIR.log
# ps -ef |grep lirc
root     18132     1  0 09:27 ?        00:00:00 /usr/sbin/lircd --output=/run/lirc/lircd --device=/dev/lirc0
# lircd -H ? 2>&1 | grep igua
	iguanaIR
# ls -l /dev/lirc*
lrwxrwxrwx 1 root root 15 Apr 27 09:27 /dev/lircd -> /run/lirc/lircd
# igclient --get-version
get version: success: version=0x0308
```

But still, sending commands doesn't work:
```
$ irsend SET_TRANSMITTERS 1
irsend: command failed: SET_TRANSMITTERS 1
irsend: hardware does not support sending
```

At this point I'm stumped.

IguanaBen, 2017-04-27:
Can you post your LIRC configuration? On the command line, you want lircd you want to set the driver to be our driver. Something like
```
lircd --driver=iguanaIR
```

depending on the distro, this can often be set in /etc/lirc/hardware.conf

brianjmurrell, 2017-05-01:
> Can you post your LIRC configuration?

Updated configuration:
```
REMOTE="Iguanaworks USB IR Transceiver"
REMOTE_MODULES="None"
REMOTE_DRIVER=""
REMOTE_DEVICE="/dev/lirc0"
REMOTE_SOCKET=""
REMOTE_LIRCD_CONF=""
REMOTE_LIRCD_ARGS=""
DRIVER="iguanaIR"
DEVICE="/dev/iguanaIR/0"

TRANSMITTER="My IguanaIR custom config"
TRANSMITTER_MODULES=""
TRANSMITTER_DRIVER="iguanaIR"
TRANSMITTER_DEVICE="/dev/iguanaIR/0"
TRANSMITTER_SOCKET=""
TRANSMITTER_LIRCD_CONF="cisco/DTA50"
TRANSMITTER_LIRCD_ARGS=""

DISABLE_KERNEL_SUPPORT="true"
START_LIRCD="true"
LOAD_MODULES="false"
LIRCMD_CONF=""
FORCE_NONINTERACTIVE_RECONFIGURATION="false"
START_LIRCMD=""

```

> On the command line, you want lircd you want to set the driver to be our driver.

I now have this:

```
# ps -ef | grep lirc
root     17011     1  0 10:55 ?        00:00:00 /usr/sbin/lircd --output=/run/lirc/lircd1 --driver=iguanaIR --device=/dev/iguanaIR/0 --listen --pidfile=/run/lirc/lircd1.pid
root     17013     1  0 10:55 ?        00:00:00 /usr/sbin/lircd --output=/run/lirc/lircd --device=/dev/lirc0 --connect=localhost 8765
root     17015 17011  0 10:55 ?        00:00:00 /usr/sbin/lircd --output=/run/lirc/lircd1 --driver=iguanaIR --device=/dev/iguanaIR/0 --listen --pidfile=/run/lirc/lircd1.pid
```

But still getting:
```
$ irsend SET_TRANSMITTERS 1
irsend: command failed: SET_TRANSMITTERS 1
irsend: hardware does not support sending
```

IguanaBen, 2017-05-01:
It looks like you have two instance of LIRC running (/run/lirc/lircd1 and /run/lirc/lircd). When you run irsend it isn't clear which lirc you are talking to and only one is using the lirc driver. Can you restart lirc (or maybe the machine) or just kill the lirc processes and start a fresh instance of lirc and see if that fixes it?

IguanaBen, 2017-05-01:
You can also try
"""
irsend --device=/run/lirc/lircd1 SET_TRANSMITTERS 1
"""
to try sending to the right lirc instance.

leamas, 2017-05-02:
Drive-by comment: The two instances of lircd uses different drivers and this is likely to cause problems. The receiving instance  on /dev/lirc0 will use the kernel driver, while the sending instance on  /dev/iguanaIR/0 uses the iguanaIR user-space driver. But these cannot use the the same device, there will be collisions on the physical device access (should be visible in dmesg). 

The best ir perhaps to use a single lircd instance for both sending and receiving. Which driver to use (kernel or iguanaIR) is hard to say, basically should any of them be fine.

BTW, the /dev/lircd link adds nothing but confusion. The /dev directory is maintained by udev, and manually added links here is basically a Bad Thing. Just configure the applications use the proper socket, normally /run/lirc/lircd for apps consuming IR events.

brianjmurrell, 2017-06-20:
The thing is, this is the result of every single upgrade I do on Ubuntu.  Without fail, after every single update/upgrade I do, I have to spend hours dicking around with LIRC and the Iguanaworks IR config.  Why is that?  Why can't the packages in your repo just work with upgrades?  Why do I have to rebuild lirc every time ubuntu rebuilds it also?  Why can't the [repo](http://iguanaworks.net/downloads/debian) that you supply, just have everything I need, including an lirc package that will always override the one provided by Ubuntu (so, yes, you have to rebuild every time they rebuild).  Or why can't you upstream your work to Ubuntu so that nobody has to rebuild the lirc package because the one that comes from Ubuntu will have your stuff in it?  This is all very frustrating.

Also, why is none of this configurable with `dpkg-reconfigure lirc`?

When I run that the _Iguanaworks USB IR Transceiver_ is in the list of _Remote control configuration_ but when I choose that I don't see anything in the subsequent _IR Transmitter_ list that has anything to do with the Iguanaworks transceiver, at which point `dpkg-reconfigure lirc` becomes useless and I end up having to manually build lirc configuration files by hand.  This is not very user-friendly **at all**.  My mom would never be able to do any of this.

All of this is ultimately is why I have multiple lirc configurations running at the same time.

The ongoing experience with your hardware and lack of cohesive software support makes me reluctant (reluctant enough that I actually don't recommend) to recommend your hardware to anyone looking for an IR solution.  That is not a good place for you to be in as a business.

leamas, 2017-06-20:
Hi!

There are really two things here: lirc and the iguanair lirc driver.

As for the driver, this is a question for the IgiuanaIR folks. 

However, as for lirc, there is actually some progress in this respect. The zesty version of lirc does *not* need to be re-compiled to use the igaunair driver. The "new" lirc version (used in other distros since several years) uses dynamic driver loading. Practically, this means that it's possible to just compile the driver (which is simple) instead of re-compiling lirc.

There is a new version under way of the iguanaIR driver. Once this is completed (perhaps Real Soon, dunno) I plan to create a debian package for it. This means that installing the driver just will be to install the lirc-drv-iguanair package. Until then, you could just compile the driver yourself.

I'm sorry to hear about your experiences, but I can understand them. The core issue here is that the Debian/Ubuntu packagers haven't used a modern lirc but a ancient 0.9.0 version; this has created this mess for the IguanaIR devs. It's even worse since they do not include iguanair in their ancient builds. E. g., fedora has always supported iguanair directly, even when using lirc 0.9.0. So, your experience is not really about IguanaIR, but the combination IguanaIR + Ubuntu.

And even for Ubuntu, there is hope :)

My personal experiences are for the Fedora/IguanaIR combination which I recommend form time to time (as well as some other solutions supported by more recent lirc). Actually, despite the Debian/ubuntu shortcomings, IguanaIR is one of the best options with respect to lirc support.

Finally, note that the responsible for failing upgrades really are the downstream packagers (Debian/Ubuntu). There is not much upstream developers like IguanaIR can do about this.  The only reason there is a separate IguanaIR Ubuntu PPA  is, IMHO,  the  "sub-optimal" debian/ubuntu packaging.  

Disclaimer: After hesitating a loooong time, I'm now maintaining modern lirc packages for Debian/Ubuntu.

jdunn14, 2017-06-20:
I'll weigh in as well and say that at this point if you are using our hardware the way most people do, for send and receive, there is not a great reason to use the igdaemon/igclient driver on Ubuntu.  If you uninstall the iguanaIR packages that should remove the /etc/modprobe.d/iguanair.conf file that blackllists the kernel driver.  With that gone then every time you boot or plug in the hardware the iguanair kernel driver loads and LIRC can connect directly with that using it's default driver.  I did notice that in some versions you have to explicitly pass "-H default" to lircd but I  doubt that's necessary now.

For the majority of people the combination of the kernel driver and LIRC just works.

leamas, 2017-06-20:
@judunn: Indeed. I can nothing but concur (why didn't I write that myself?)

brianjmurrell, 2017-06-21:
OK.  So I have removed the _iguanair_ package:

```
# apt-get remove iguanair
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  iguanair
0 upgraded, 0 newly installed, 1 to remove and 114 not upgraded.
After this operation, 389 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 506854 files and directories currently installed.)
Removing iguanair (1.1.0) ...
 * Stopping Iguanaworks USB IR daemon...                                                          [ OK ] 
```
That did not remove the `/etc/modprobe.d/iguanair.conf` so I removed it manually:

```
# ls -l /etc/modprobe.d/iguanair.conf 
-rw-r--r-- 1 root root 19 Apr 24  2015 /etc/modprobe.d/iguanair.conf
# rm -f /etc/modprobe.d/iguanair.conf 
```

and installed the _iguanair_ module:

```
# modprobe iguanair
# dmesg
...
[4759441.734541] Registered IR keymap rc-rc6-mce
[4759441.734617] input: IguanaWorks USB IR Transceiver version 0x0308 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.4/3-2.4:1.0/rc/rc0/input16
[4759441.746320] rc0: IguanaWorks USB IR Transceiver version 0x0308 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.4/3-2.4:1.0/rc/rc0
[4759441.754803] IR RC5(x) protocol handler initialized
[4759441.756104] IR RC6 protocol handler initialized
[4759441.763043] IR Sony protocol handler initialized
[4759441.766902] IR MCE Keyboard/mouse protocol handler initialized
[4759441.768743] lirc_dev: IR Remote Control driver registered, major 250 
[4759441.770652] input: MCE IR Keyboard/Mouse (iguanair) as /devices/virtual/input/input17
[4759441.772563] usbcore: registered new interface driver iguanair
[4759441.773008] rc rc0: lirc_dev: driver ir-lirc-codec (iguanair) registered at minor = 0
[4759441.773010] IR LIRC bridge handler initialized
[4759441.797776] IR NEC protocol handler initialized
[4759441.797868] IR SANYO protocol handler initialized
[4759441.804872] IR JVC protocol handler initialized
```

When I restart _lirc_ I get an error:

```
# service lirc restart
 * Stopping remote control daemon(s): LIRC                                                        [ OK ] 
 * Starting remote control daemon(s) :                                                                   /usr/sbin/lircd: error while loading shared libraries: libiguanaIR.so: cannot open shared object file: No such file or directory
```

Maybe my `hardware.conf` needs adjusting to use the kernel module instead of the userspace module as I was doing?

```
# /etc/lirc/hardware.conf
#
#Chosen Remote Control
REMOTE="None"
REMOTE_MODULES=""
REMOTE_DRIVER=""
REMOTE_DEVICE=""
REMOTE_SOCKET=""
REMOTE_LIRCD_CONF=""
REMOTE_LIRCD_ARGS=""
DRIVER="iguanaIR"
DEVICE="/dev/iguanaIR/0"

#Chosen IR Transmitter
TRANSMITTER="My IguanaIR custom config"
TRANSMITTER_MODULES=""
TRANSMITTER_DRIVER="iguanaIR"
TRANSMITTER_DEVICE="/dev/iguanaIR/0"
TRANSMITTER_SOCKET=""
TRANSMITTER_LIRCD_CONF="cisco/DTA50"
TRANSMITTER_LIRCD_ARGS=""

#Disable kernel support.
#Typically, lirc will disable in-kernel support for ir devices in order to
#handle them internally.  Set to false to prevent lirc from disabling this
#in-kernel support. 
DISABLE_KERNEL_SUPPORT="true"

#Enable lircd
START_LIRCD="true"

#Don't start lircmd even if there seems to be a good config file
#START_LIRCMD="false"

#Try to load appropriate kernel modules
LOAD_MODULES="false"

# Default configuration files for your hardware if any
LIRCMD_CONF=""

#Forcing noninteractive reconfiguration
#If lirc is to be reconfigured by an external application
#that doesn't have a debconf frontend available, the noninteractive
#frontend can be invoked and set to parse REMOTE and TRANSMITTER
#It will then populate all other variables without any user input
#If you would like to configure lirc via standard methods, be sure
#to leave this set to "false"
FORCE_NONINTERACTIVE_RECONFIGURATION="false"
START_LIRCMD=""
```

To be clear, I am using this hardware to "blast IR", not to receive commands from a remote.

brianjmurrell, 2017-06-21:
Now that I go back and [read](http://www.iguanaworks.net/wiki/doku.php?id=usbir:gettingstarted) it says:

>There are a some common reasons to configure LIRC instead:
>
> ...
> 2. You plan to use the sending (blasting) capabilities of our device; this requires LIRC.

I'm guessing that negates the advise in this ticket to use the kernel module, is that right?

brianjmurrell, 2017-06-21:
OK.  So I went back to the beginning and re-did all of the [steps](http://www.iguanaworks.net/wiki/doku.php?id=usbir:gettingstarted) and it's now working again.

Hopefully it survives the next update/upgrade.

closed 2017-06-21 by brianjmurrell

jdunn14, 2017-06-21:
As stated above "if you are using our hardware the way most people do, for send and receive, there is not a great reason to use the igdaemon/igclient driver on Ubuntu".  Both send (blasting) and receive are supported through the kernel driver, but if you've got it working just leave it be.

brianjmurrell, 2017-06-21:
Does [this page](http://www.iguanaworks.net/wiki/doku.php?id=usbir:gettingstarted) need updating then where it says:

> There are a some common reasons to configure LIRC instead:
> 
> ...
> 2. You plan to use the sending (blasting) capabilities of our device; this requires LIRC.

leamas, 2017-06-21:
Yes, But that discussion just refers to using lirc versus not using it at all. To blast, you need lirc. But you don't need the iguanair driver, the kernel driver + lirc's 'default' driver (silly name) is just fine.

nahaff, 2023-07-23:
A little more about moving from the iguanair daemon to the builtin iguanair driver would have been helpful when I did it. Something like the following.

Use your package manager to remove the iguanair package.
Delete any of those conf files that are blacklisting the default driver.
grep iguana /etc/modprobe.d/*.conf
I had two:
/etc/modprobe.d/60-blacklist-kernel-iguanair.conf
/etc/modprobe.d/iguanair.conf
In LIRC, driver should be explicitly set to default. Mine was set to iguanair, and what most people would consider default (the settings out of the box), is devinput for Ubuntu. I made that change by modifying lirc_options.conf, but they could probably also be set as command line parameters in service files or other configuration files.
Also set device to auto in the same area.
Reboot.