iguanaworks/iguanair #41: Error: couldn't open connection to iguanaIR daemon: No such file or directory

fcueto, 2017-06-24:
Hi,

using debian stretch, LIRC 0.9.4c and the iguanair compiled driver, I run into the following issue.

The iguaraIR sockets are properly created:

```
root@light:~# ll /var/run/iguanaIR/
total 0
srwxrwxrwx 1 iguanair iguanair 0 Jun 24 23:54 0
srwxrwxrwx 1 iguanair iguanair 0 Jun 24 23:54 ctl
lrwxrwxrwx 1 iguanair iguanair 1 Jun 24 23:54 iguana -> 0
```

I can verify that IguanaIR is properly working using igclient:
```
root@light:~# igclient --get-version
get version: success: version=0x0308
```

but lircd complains when I want to send IR signals:

```
Jun 25 00:53:30 light lircd-0.9.4c[9016]: Notice: accepted new client on /var/run/lirc/lircd
Jun 25 00:53:30 light irsend[9025]: lirc_command_run: Sending: send_once SONY KEY_POWER
Jun 25 00:53:30 light lircd-0.9.4c[9016]: Error: couldn't open connection to iguanaIR daemon: No such file or directory
Jun 25 00:53:30 light lircd-0.9.4c[9016]: Warning: Failed to initialize hardware
Jun 25 00:53:30 light lircd-0.9.4c[9016]: Error: error processing command: send_once SONY KEY_POWER
Jun 25 00:53:30 light lircd-0.9.4c[9016]: Error: transmission failed
```

I found out using strace that lircd tries to open this socket : /var/run/iguanaIR/auto

The issue can be resolved by creating a symlink /var/run/iguanaIR/auto targeting /var/run/iguanaIR/0

Why is lircd trying to open that file "auto" ?

fcueto, 2017-06-25:
I found out that the file /etc/lirc/lirc_options.conf needed to be edited:

from

```
driver          = devinput
device          = auto
```

to

```
driver          = iguanaIR
device          = /var/run/iguanaIR/0
```

Strange that it worked in 0.9.4c but not in 0.9.4d without this change

jdunn14, 2017-06-26:
Thanks for reporting this.  I think at some point in the lirc iguanair driver development we lost the name of the default device.  It should default to "0" but instead changed to "auto" as you saw.  Instead of using /var/run/iguanaIR/0 you could probably use "0" or if you set an id for the device you could use that name.

Thanks again, and we'll make sure this mismatch gets cleared up in the next release.

jdunn14, 2017-07-26:
I committed a fix for this.  Basically if the lirc driver sees "auto" it calls iguanaConnect with a NULL to signify "just pick the first one".

closed 2017-07-26 by jdunn14