iguanaworks/iguanair #44: Support 64 bit on Windows

jdunn14, 2017-08-02:
We should be able to build the service as 64 bit, although it doesn't really matter.  However, we'll want to supply both 32 and 64 bit versions of the various libraries.  The pipe layer between the client library and the service means we can handle clients of either architecture, but we'll want to test to make sure there isn't a stray "int" laying around.

kossatzd, 2018-03-14:
just replaced my one PC that was using an iguanaIR transceiver.  It's now a Win 7 64 bit machine.  Can't get it working with WinLIRC like it was before (old machine 32 bit).  Do I need to wait for something from this project?  from WinLIRC?  Or am I forgetting something?

jdunn14, 2018-03-14:
You should have everything you need, but I'm not shocked if Win7 64 doesn't install the details properly.  We supply signed 64 bit drivers so there shouldn't be an issue on the kernel side. Unless you built WinLIRC yourself it will be 32 bit, so the libraries and igdaemon service should all be 32 bit as well.

Does the iguanaworks service exist?  Is it running?  If it is can you plug in a device and then run igclient --get-version?  You'll need to run it from a command prompt in the installation directory (defaults to c:\program files x86\IguanaIR on newer 64 bit systems).

I'm happy to try and walk you through getting this working, but we will not be making changes to the release to support anything Win7 specific.

kossatzd, 2018-03-16:
...oh my.  i went to check igclient and realized the installer was just sitting in my Downloads folder, never actually ran it.  Could have sworn i did.  No issues once I actually installed the service.  Thanks!

jdunn14, 2018-03-16:
Ha!  Well, glad that's resolved.  I wasn't sure if the installer would work properly on 64 bit Win7 but nothing specific comes to mind that would cause a problem.  Enjoy your hardware.