iguanaworks/iguanair #50: Crashing on Ubuntu 20.04

morganchristiansson, 2020-08-13:
Latest code in git crashes on Ubuntu 20.04

The latest tarball (instead of git master) does not crash but lirc-drv-iguanair doesn't link and lirc-lsplugins will say ```Error: /usr/lib/x86_64-linux-gnu/lirc/plugins/iguanair.so: undefined symbol: iguanaListDevices```

```gdb igdaemon
(gdb) r --devices
Starting program: /usr/bin/igdaemon --devices
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7fc6c00 in findDeviceEndpoints (info=0x4, maxPacketSize=0x7ffff7faa012) at /home/morgan/Downloads/iguanair/software/usb_ir/drivers/libusb.c:576
warning: Source file is more recent than executable.
576         usbDevice *head, *prev = NULL;
```

Not sure what the problem is..

jnbiesboer, 2020-09-28:
I attempted the installation on Raspberry Pi OS 08-20-2020.  From GitHub (software/lirc-drv-iguanair/).  Got everything to work in the "Getting Started" instructions [here](https://www.iguanaworks.net/wiki/doku.php?id=usbir:gettingstarted) until Step 4.  After Step 4: `sudo make install`,  `lirc-lsplugins -q` reports the same error about the "undefined symbol".

morganchristiansson, 2021-01-28:
@jnbiesboer On a Raspberry Pi you can connect a IR LED directly to GPIO pins https://www.hackster.io/austin-stanton/creating-a-raspberry-pi-universal-remote-with-lirc-2fd581

Just need an IR led, resistor and transistor.

tomjuggler, 2022-11-15:
I have the same error message on Ubuntu 22.04 after compiling lirc-drv-iguanair using "`sudo make install`".